nehilo/Klipper-KingRoon-Printers #9: UART mode solder jumpers

Upcycle-Electronics, 2022-08-28:
Hello,

Why do you list using the pins on one side of the board to setup UART mode here: https://github.com/nehilo/Klipper-KingRoon-Printers#tmc-uart-pin-kingroon-13-motherboard--for-advanced-users-

I have read the [TMC2225 datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2225_datasheet_rev1.14.pdf). The box at the top of page 16 says, "The UART line must be logic high during idle state. Therefore, the power down function cannot be assigned by the pin PDN_UART in between transmissions." The 20k resistor (marked "30C") for the TMC2225 pin 17 (PDN_UART) is connected to ground. In other words, the resistor must be removed to enable UART mode just like you mentioned. 

The part that I do not understand is why you are not using the jumper pins on the soldering pads near each of the 20k resistors. These jumper pads are connected to a GD32F303V pin and to each TMC2225 pin 17 (PDN_UART). 

- X Axis = p30 / PA5
- Y Axis = p7 / PC13
- Z Axis = p64 / PC7
- E0 Axis = p68 / PA9
- E1 Axis = p67 / PA8

I think the hotend fan should also get moved from the JST connectors with 24V power to the HE1 screw terminals. This will enable the fan to be turned on only when the hotend is hot or a print is running. It will also enable users to tune the fan speed with PWM and perhaps run the fan quieter.

I would appreciate full board schematics to avoid the need to reverse engineer more of this. I would also like to get the screen working with Klipper. I need the documentation like what kind of driver this uses or what Makerbase screen module version/code this is based on.

[image]

nehilo, 2022-08-28:
Know about this, need time to upgrade manual

Hotend fan control u can chk in configuration with pin

closed 2022-09-11 by nehilo

odaki, 2022-09-28:
@Upcycle-Electronics 
Thank you for the very useful information.
> X Axis = p30 / PA5
Y Axis = p7 / PC13
Z Axis = p64 / PC7
E0 Axis = p68 / PA9
E1 Axis = p67 / PA8

I have checked my V1.3 board and it appears that the pin assignments are slightly different for the E axis.
Could you please confirm this?
E0 Axis = p69 / PA10
E1 Axis = p68 / PA9

Thank you.