nehilo/Klipper-KingRoon-Printers #15: Having issue on 3D touch "BLTouch failed to raise probe"

chaopeng, 2022-12-24:
I did follow the video to connect 2pin to Z+ and 3pin to BLT, and ensure the GND is on correct pin. I got "BLTouch failed to raise probe" error and pin_down does not get open on QUERY_PROBE.

config is same as this repo
```
[bltouch]
sensor_pin: ^PA11
control_pin: PA8
speed: 20
x_offset: 30
y_offset: -10
z_offset: 1.099
pin_move_time: 0.400
probe_with_touch_mode: False
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: True
lift_speed: 15
samples: 2
sample_retract_dist: 5.0
samples_tolerance: 0.05
samples_tolerance_retries: 5
```

```
$ BLTOUCH_DEBUG COMMAND=pin_down
// Sending BLTOUCH_DEBUG COMMAND=pin_down
$ QUERY_PROBE
// probe: TRIGGERED
$ BLTOUCH_DEBUG COMMAND=pin_up
// Sending BLTOUCH_DEBUG COMMAND=pin_up
$ QUERY_PROBE
// probe: TRIGGERED
```

chaopeng, 2022-12-24:
I think the Z+ is PC4, change the config to sensor_pin: ^PC4, got always open

nehilo, 2022-12-24:
> I think the Z+ is PC4, change the config to sensor_pin: ^PC4, got always open
> 
> 

Yep, if you want to use Z+ pin
In my cfg use Z- because with bltouch u don't need to use endstop Z

chaopeng, 2022-12-26:
I endup get my 3d touch works with

```
probe_with_touch_mode: False
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: False
```

closed 2022-12-26 by chaopeng

JacsonRitzmann, 2023-03-29:
@chaopeng  @nehilo Please, how did you solve it, how was the final settings [bltouch]?
My settings:
sensor_pin: ^PC4
control_pin: PA8
x_offset: 30
y_offset: -10
z_offset: 1
probe_with_touch_mode: False

chaopeng, 2023-03-29:
Maybe the wiring is wrong.