nehilo/Klipper-KingRoon-Printers #18: BLTOUCH is always giving me an open state

Coblenugget, 2022-12-31:
I have made sure that the wiring is correct and I have it plugged into z- where the z endstop use to be at. I have tried different cables and a clone with the same results

[bltouch]
sensor_pin: ^PA11
control_pin: PA8
speed: 20
x_offset: -25
y_offset: -28
z_offset: 1.099
pin_move_time: 0.400
probe_with_touch_mode: False
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: True
lift_speed: 15
samples: 2
sample_retract_dist: 5.0
samples_tolerance: 0.05
samples_tolerance_retries: 5

[safe_z_home]
home_xy_position: 188,181
speed: 50
z_hop: 8
z_hop_speed: 25

[image]
[image]

wonsnot, 2022-12-31:
I have the same issue with both a 3d touch from klipper and with an official bltouch.

wxt203814, 2023-01-17:
the hashtag is what made mine work infront of the " probe_touch_mode" argument 

[bltouch]
sensor_pin: ^PA11
control_pin: PA8
speed: 20
x_offset: -25
y_offset: -28
z_offset: 1.099
pin_move_time: 0.400
#probe_with_touch_mode: True # for original BLTouch v 3.0
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: True
lift_speed: 15
samples: 2
sample_retract_dist: 5.0
samples_tolerance: 0.05
samples_tolerance_retries: 5

nehilo, 2023-01-17:
> the hashtag is what made mine work infront of the " probe_touch_mode" argument
>
> [bltouch] sensor_pin: ^PA11 control_pin: PA8 speed: 20 x_offset: -25 y_offset: -28 z_offset: 1.099 pin_move_time: [...]

"probe_touch_mode" argument need for another bltouch clone. Some of it work without "probe_touch_mode"

closed 2023-03-07 by nehilo