nehilo/Klipper-KingRoon-Printers #21: KP5L: wrong tmc uart_pins?

Piezoid, 2023-06-04:
KingRoon is referring to your repository in their [KP5L Klipper Guide](https://kingroon.com/es/blogs/downloads/how-to-set-up-klipper-on-kingroon-kp5l-3d-printers), and many comments reports communication issues with TMC drivers.

Looking though Marlin's [configs](https://github.com/MarlinFirmware/Configurations/blob/import-2.1.x/config/examples/Kingroon/KP5L/Configuration.h#LL72C25-L72C25) and [board definitions](https://github.com/MarlinFirmware/Configurations/blob/import-2.1.x/config/examples/Kingroon/KP5L/Configuration.h#LL72C25-L72C25), it seems to have the drivers them configured in non-standalone mode and communicating with them through PD5/PD7/PD4/PD9 for X/Y/Z/E.

Could there possibly be different board versions for this printer?

I don't have a KP5L for testing, I was simply trying to help a [user](https://klipper.discourse.group/t/cannot-get-klipper-to-connect-to-my-new-kingroon-kp5l/8717/19), but they haven't reported if these pin settings works.

nehilo, 2023-06-04:
not issue chk readme before open https://github.com/nehilo/Klipper-KingRoon-Printers/tree/main#tmc-uart-pin-kingroon-motherboard--for-advanced-users-

closed 2023-06-04 by nehilo

Piezoid, 2023-06-04:
I read this, as the other issues for the KP3S `uart_pins`.
But unlike the KP3S, marlin seems to be using the driver in non-standalone mode. Also unlike the KP3S, your KP5L doesn't have the `# disable to use stock modification` on the `tmc.cfg` include line.